mjbots/moteus #2: FAILED: Build did NOT complete successfully

rituraj-01, 2020-10-10:
We have bought the moteus r4.3 developer kit. Following the video and github tutorial, I'm getting the compilation error.
I'm following [](https://github.com/mjbots/moteus/blob/master/docs/getting_started.md), 
`On running this tools/bazel test -c opt --cpu=stm32g4 //:target`, it compiles with some downloading errors, but I fixed them by manually downloading the files.
but on running `tools/bazel test --cpu=k8 //:host`, I'm I'm getting the following error

[image]

[Error-1.txt](https://github.com/mjbots/moteus/files/5359153/Error-1.txt)

`sacgc@sacgc-VirtualBox:~/Documents/moteus$ tools/bazel test --cpu=k8 //:host
INFO: Analyzed 3 targets (0 packages loaded, 0 targets configured).
INFO: Found 3 test targets...
ERROR: /home/sacgc/Documents/moteus/moteus/tool/BUILD:38:11: C++ compilation of rule '//moteus/tool:moteus_tool_lib' failed (Killed): clang-clang failed: error executing command external/com_github_mjbots_rules_mbed/tools/cc_toolchain/wrapper/clang-clang -MD -MF bazel-out/k8-opt/bin/moteus/tool/_objs/moteus_tool_lib/moteus_tool.d ... (remaining 79 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox clang-clang failed: error executing command external/com_github_mjbots_rules_mbed/tools/cc_toolchain/wrapper/clang-clang -MD -MF bazel-out/k8-opt/bin/moteus/tool/_objs/moteus_tool_lib/moteus_tool.d ... (remaining 79 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
INFO: Elapsed time: 114.111s, Critical Path: 62.59s
INFO: 17 processes: 17 linux-sandbox.
FAILED: Build did NOT complete successfully
//moteus:dummy_host_test                                              NO STATUS
//moteus:test                                                         NO STATUS
//moteus/tool:test                                              FAILED TO BUILD

FAILED: Build did NOT complete successfully
`

I'm running ubuntu 18.04 on Virtual Machine.
Please help

jpieper, 2020-10-10:
It says `(Killed)` which makes me think maybe you ran out of memory or disk space.  Can you run `dmesg` and see if the kernel reported an out of memory error?

rituraj-01, 2020-10-10:
Yes it is showing out of the memory. do you have some suggestions for configuration on Virtual Machine. I have given it 2GB of RAM access.

jpieper, 2020-10-10:
I don't know exactly how much is necessary.  I would expect you should have at least 4G.

jpieper, 2020-10-10:
Were you able to get it working?

rituraj-01, 2020-10-10:
There was one more problem occurred, related to some package. If I fail to solve it, I'll need this thread.

rituraj-01, 2020-10-11:
Whie Running the `./bazel-out/k8-opt/bin/moteus/tool/tview --devices=1` command. I'm getting error 
```
Falling back to PySide1
Gtk-Message: 10:25:41.876: Failed to load module "canberra-gtk-module"
Traceback (most recent call last):
  File "/home/sacgc/Documents/moteus/./bazel-out/k8-opt/bin/moteus/tool/tview.runfiles/com_github_mjbots_moteus/moteus/tool/tview.py", line 1154, in _handle_user_input
    device_nums = [self.devices[0].number]
IndexError: list index out of range
```

[image]
[image]

rituraj-01, 2020-10-11:
I solved the module problem (Gtk-Message: 10:25:41.876: Failed to load module "canberra-gtk-module") by installing following modules:
`sudo apt install libcanberra-gtk-module libcanberra-gtk3-module`

but it still shows:
`Falling back to PySide1`
and doesn't loads the GUI.

jpieper, 2020-10-11:
My guess is that you're having problems passing through USB to your VirtualBox.  Can you verify that `/dev/ttyACM0` is present there?

Also, I can help you faster/in-real-time if you're willing to join discord: https://discord.gg/W4hUpBb  The #moteus channel is a fine place to ask questions.

rituraj-01, 2020-10-11:
Thank you @jpieper I used a different virtual machine and it worked.

closed 2020-10-11 by rituraj-01