mjbots/moteus #7: Windows 10, tview console err unknown name.. Ubuntu "No such device"

ragemaster, 2021-02-01:
under Windows 10, tview console err unknown name.. and no telemetry data however in the console is shows readings from the device.. nothing else happens as demonstrating in your getting started video..
Tried to get it to work in Ubuntu 20.04 and the terminal output says this:
File "/home/terrence/.local/lib/python3.8/site-packages/moteus_gui/tview.py", line 856, in _make_transport
    return moteus.get_singleton_transport(self.options)
  File "/home/terrence/.local/lib/python3.8/site-packages/moteus/moteus.py", line 120, in get_singleton_transport
    raise RuntimeError("Unable to find a default transport, tried: {}".format(
RuntimeError: Unable to find a default transport, tried: (<moteus.moteus.FdcanusbFactory object at 0x7fd01ef87a90>, "[Errno 13] could not open port /dev/serial/by-id/usb-mjbots_fdcanusb_299C5406-if00: [Errno 13] Permission denied: '/dev/serial/by-id/usb-mjbots_fdcanusb_299C5406-if00'"),(<moteus.moteus.PythonCanFactory object at 0x7fd01ef87ac0>, '[Errno 19] No such device')

jpieper, 2021-02-01:
1) For Ubuntu, that error message indicates that you do not have permission to read the USB serial device.  You could try installing the `70-fdcanusb.rules` following the instructions at the top of the file: https://github.com/mjbots/fdcanusb/blob/master/70-fdcanusb.rules

2) For Windows, It is unclear exactly what error message you got.  Did you install `moteus-gui`, then were unable to type `tview` at a command prompt because it said no such command or file name?  You may have to run the program as `python3 -m moteus_gui.tview` if you didn't configure python to put user scripts in your system path.

ragemaster, 2021-02-02:
still cant get windows 10 running however the for linux udev rules works! now my wheels can spin, Thanks

jpieper, 2021-02-02:
Great!  If you want to get Windows working, come over to the discord where we can help you interactively: https://discord.gg/W4hUpBb

closed 2021-02-02 by jpieper