mjbots/moteus #21: When I send a cmd to mjbots via fdcan, mjbots only responds the first cmd and no message reply.

Mu-feng, 2022-01-21:
When I use FDCAN command to control the motor, it only responds to the first command and does not reply.Then if I send a command again and again, it has a very small accumulation of responses, with no reply.
I observered the fdcan-cmd that i have sent, and found it is corrected.
So, I am very conffused about this scene.
Do you have any solution for this?

jpieper, 2022-01-21:
We can help you debug, although it may take a few back and forths.  Are you able to join the mjbots Discord at: https://discord.gg/W4hUpBb  as real-time back and forth will be faster.

If not, the first question is:

First, what CAN transceiver are you using and what software are you using send and receive CAN commands?

Mu-feng, 2022-01-21:
What I used is BUSMASTE USB-CAN and its software is BUSMASTE.
I send this two cmd for different torque, one of  is "01 00 09 06 1C 01 00 00 00 14 04 00", and another is "01 00 08 06 1C 20 00 00 00 14 04 00".
Final, thanks for your help and I have registered Mjbots Discord as you said.

jpieper, 2022-01-21:
Great, can you post a message in the #moteus channel and we can start debugging?

Mu-feng, 2022-01-21:
Ok, thank you.
I will post it later.

jpieper, 2022-02-16:
Closing this as I believe it was resolved out of band.

closed 2022-02-16 by jpieper