mjbots/moteus #22: dependency-conflicts PySide2

ator1811, 2022-02-16:
Hi, I'm using Python 3.10.2
When installing moteus with pip3 "install moteus_gui" I get the following error:
`Collecting moteus_gui
  Using cached moteus_gui-0.3.32-py3-none-any.whl (12 kB)
Collecting asyncqt>=0.8
  Using cached asyncqt-0.8.0.tar.gz (12 kB)
  Preparing metadata (setup.py) ... done
Collecting moteus_gui
  Using cached moteus_gui-0.3.31-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.30-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.29-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.28-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.27-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.26-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.25-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.24-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.23-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.22-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.21-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.20-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.19-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.18-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.17-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.16-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.15-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.14-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.13-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.12-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.4-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.3-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.2-py3-none-any.whl (12 kB)
  Using cached moteus_gui-0.3.0-py3-none-any.whl (12 kB)
ERROR: Cannot install moteus-gui==0.3.0, moteus-gui==0.3.12, moteus-gui==0.3.13, moteus-gui==0.3.14, moteus-gui==0.3.15, moteus-gui==0.3.16, moteus-gui==0.3.17, moteus-gui==0.3.18, moteus-gui==0.3.19, moteus-gui==0.3.2, moteus-gui==0.3.20, moteus-gui==0.3.21, moteus-gui==0.3.22, moteus-gui==0.3.23, moteus-gui==0.3.24, moteus-gui==0.3.25, moteus-gui==0.3.26, moteus-gui==0.3.27, moteus-gui==0.3.28, moteus-gui==0.3.29, moteus-gui==0.3.3, moteus-gui==0.3.30, moteus-gui==0.3.31, moteus-gui==0.3.32 and moteus-gui==0.3.4 because these package versions have conflicting dependencies.

The conflict is caused by:
    moteus-gui 0.3.32 depends on PySide2
    moteus-gui 0.3.31 depends on PySide2
    moteus-gui 0.3.30 depends on PySide2
    moteus-gui 0.3.29 depends on PySide2
    moteus-gui 0.3.28 depends on PySide2
    moteus-gui 0.3.27 depends on PySide2
    moteus-gui 0.3.26 depends on PySide2
    moteus-gui 0.3.25 depends on PySide2
    moteus-gui 0.3.24 depends on PySide2
    moteus-gui 0.3.23 depends on PySide2
    moteus-gui 0.3.22 depends on PySide2
    moteus-gui 0.3.21 depends on PySide2
    moteus-gui 0.3.20 depends on PySide2
    moteus-gui 0.3.19 depends on PySide2
    moteus-gui 0.3.18 depends on PySide2
    moteus-gui 0.3.17 depends on PySide2
    moteus-gui 0.3.16 depends on PySide2
    moteus-gui 0.3.15 depends on PySide2
    moteus-gui 0.3.14 depends on PySide2
    moteus-gui 0.3.13 depends on PySide2
    moteus-gui 0.3.12 depends on PySide2
    moteus-gui 0.3.4 depends on PySide2
    moteus-gui 0.3.3 depends on PySide2
    moteus-gui 0.3.2 depends on PySide2
    moteus-gui 0.3.0 depends on PySide2

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
`
installed packages (pip3 list):
Package    Version
---------- -------
future     0.18.2
iso8601    1.0.2
pip        22.0.3
PyYAML     6.0
serial     0.0.97
setuptools 58.1.0

What is going wrong?
Thanks for your help

jpieper, 2022-02-16:
What operating system and version are you using?

ator1811, 2022-02-16:
MacOS Monterey
v12.1 on MacBook

jpieper, 2022-02-16:
Are you on a M1 hardware?

ator1811, 2022-02-16:
yes

jpieper, 2022-02-16:
OK.  Apparently pyside2 is unlikely to ever be supported on M1.  Fixing this will require rewriting the tools to use pyside6.  I'll update this ticket accordingly, although it may be months.

In the meantime, you can use Ubuntu from a VM/Virtualbox or an Intel Mac.

ator1811, 2022-02-16:
ok, thanks. I'm glad I didn't do anything wrong. I have an Intel Mac available to use. Thanks for your response

jpieper, 2022-02-17:
For reference purposes, updating to PySide6 is not all that hard.   However, it seems that PySide6 is not available at all on the Raspberry Pi (and isn't packaged by Debian/Ubuntu either in any version, although pip does have it there).

I'll have to see if there is some way to make wheel dependencies be conditional on platform.

jpieper, 2022-02-17:
@ator1811 would you be willing to try this beta package on your M1 Mac?

https://drive.google.com/file/d/1etnpGGaj-rtROLj03Ij2kujZ6HFPQXsO/view?usp=sharing

It should use PySide6

jpieper, 2022-05-29:
Fully resolved in https://github.com/mjbots/moteus/commit/446a41bcee41eae189ed43a75b3a6d26a7ce1347

closed 2022-05-29 by jpieper