mjbots/moteus #28: how can use

ou1443989517, 2022-05-17:
I want to directly call the code based on moteus to control and query the information of the motor, how can I do it？

jpieper, 2022-05-17:
Are you asking how to use the python bindings?  Have you seen the example in: https://github.com/mjbots/moteus/blob/main/lib/python/examples/simple.py

ou1443989517, 2022-05-17:
> 您是在问如何使用 python 绑定？您是否看过以下示例：[https ://github.com/mjbots/moteus/blob/main/lib/python/examples/simple.py](https://github.com/mjbots/moteus/blob/main/lib/python/examples/simple.py)

Thank you for your reply, yes I have seen it, but the code inside simple will stop the running motor, I also used the method inside https://pypi.org/project/moteus/ like pr. position = moteus.INT16
c = moteus.Controller(position_resolution=pr) This seems to be an address printed out using print, not the desired value

ou1443989517, 2022-05-17:
> 您是在问如何使用 python 绑定？您是否看过以下示例：[https ://github.com/mjbots/moteus/blob/main/lib/python/examples/simple.py](https://github.com/mjbots/moteus/blob/main/lib/python/examples/simple.py)
e.g:<memory at 0x7ff7dd09dac0>

jpieper, 2022-05-17:
I'm sorry, but am unable to understand the problem you have.  Could you please share a full script, and exactly what you expect it to do or print out?

Are you saying if you run `simple.py` it prints out addresses?  Local testing does not do that.

ou1443989517, 2022-05-18:
Sorry, there is something wrong with my expression, here is my code
`parser_test= argparse.ArgumentParser()
max_receive_bytes=48, 
args = argparse.Namespace(target=[], can_prefix=0, verbose=False,
                          tel_stop=False, no_tel_stop=False, fdcanusb=None,
                          fdcanusb_debug=None, can_iface=None, can_chan=None,
                          force_transport=None, stop=False, info=False,
                          console=False, dump_config=False, write_config=None,
                          flash=None, no_restore_config=False,
                          bootloader_active=False, calibrate=True, cal_invert=False,
                          cal_bw_hz=100.0, encoder_bw_hz=None, cal_no_update=False,
                          cal_ll_encoder_voltage=None, cal_ll_encoder_speed=1.0,
                          cal_ll_resistance_voltage=None, cal_ll_kv_voltage=None,
                          cal_power=None, cal_speed=None, cal_voltage=None,
                          cal_kv_voltage=None, cal_motor_power=5.0,
                          cal_motor_speed=6.0, cal_max_remainder=0.1,
                          cal_raw=None, restore_cal=None, zero_offset=False)
args.devices=['1']
async def main():
    transport = moteus.Fdcanusb()
    pr=moteus.PositionResolution()
    pr.velocity=moteus.INT16
    c=moteus.Controller(position_resolution=pr)
    print(c)`
My purpose is to get the speed and torque value of the motor output like the Tview.py you wrote. I want to do something with these values

jpieper, 2022-05-18:
You need to call methods on the `Controller` instance to either send commands or make queries.

For instance:

```
async def main():
  # stuff
  c = moteus.Controller()
  result = await c.query()
  print(result)
  result = await c.set_position(position=1.0, query=True)
  print(result)
```

ou1443989517, 2022-05-18:
Cool, I will try out the code.
Well, my question is how can I use python binding to query detailed stats of the motor just like those inside the 'servo_stats' in the telemetry widget in tview?

jpieper, 2022-05-18:
The information in tview is queried using the "diagnostic" interface, which is distinct from the "register" interface.  It is not recommended for real-time use.  You can use it from python though, like so:

```
async def main():
  # stuff
  c = moteus.Controller()
  s = moteus.Stream(c)
  result = s.read_data("servo_stats")
  print(result)
```

ou1443989517, 2022-05-18:
servo: <coroutine object Stream.read_data at 0x7f7c2002e340>,This is the value I print out, which seems to be an address

jpieper, 2022-05-18:
I edited it shortly after posting to include an `await`

If you do not await an asynchronous function, it is not actually called.

```
  result = await s.read_data("servo_stats")
```

ou1443989517, 2022-05-18:
I got it! thanks
I got a few questions.
For real time use with python will dumping info like this in a loop take up too much bandwidth when I connect more motors using fdcan?  
Also, is it possible to query only certain keys of the servo_stats?

ou1443989517, 2022-05-18:
good！I can

jpieper, 2022-05-18:
As mentioned, the diagnostic interface is not intended for real time use. It both requires lots of bandwidth, and can potentially become inconsistent.  It is also undocumented, and not guaranteed to be stable across firmware versions.  It is not possible to query only individual keys.

The register interface is intended for real time use.  

https://github.com/mjbots/moteus/blob/main/docs/reference.md#a2b-registers  

What data do you want that you are not able to get from it?

ou1443989517, 2022-05-19:
Thanks, I have no more questions. Thank you for the reply.

closed 2022-05-19 by ou1443989517