mjbots/moteus #32: Tview: unexpected keyword argument

Fergus-MW, 2022-07-06:
Freshly installed tview using pip, getting this error everytime and can't interact with motor controller.
[image]

jpieper, 2022-07-06:
You need to upgrade your `moteus` library, it seems the pip version requirements aren't completely correct.

If you installed them in your user account, it would be `pip3 install --upgrade moteus`  if as root it would be `sudo pip3 install --upgrade moteus`

Fergus-MW, 2022-07-06:
Updated moteus from 0.3.16 to 0.3.42 and that fixed it. Thanks for the fast reply!

closed 2022-07-06 by Fergus-MW

jpieper, 2022-07-06:
No problem!

jpieper, 2022-07-06:
For good measure: https://github.com/mjbots/moteus/commit/9e0758333c190a2a73538bcc91ca479d91e8da34