mjbots/moteus #34: TypeError: make_rezero() got an unexpected keyword argument 'cmd'

killianstorm, 2022-08-09:
Hi,

- System: Raspberry Pi Model 4 B
- Moteus version: 0.3.43
- Type of error: TypeError

Calling the function `set_rezero` in `moteus.py` raises a `TypeError` for version `0.3.43`. 

I get the following output:
`File "/home/pi/w3_arm_research/actuator/src/controller/venv/lib/python3.9/site-packages/moteus/moteus.py", line 723, in set_rezero
    [self.make_rezero(**kwargs, cmd=cmd)]))
TypeError: make_rezero() got an unexpected keyword argument 'cmd'`

At a quick glance it seems that the passed argument `cmd=cmd` is wrongly passed. Removing it makes the function callable again.

Do I have a version mismatch or is this argument indeed wrongly passed?

Thanks!

jpieper, 2022-08-09:
Yep, that's a bug.  Should be fixed here 150d68fe953a1d8656bfe8e056e61ce87cc74575  and in version 0.3.44 on pypi

Can you confirm?

killianstorm, 2022-08-10:
All good, thanks for the quick response!

closed 2022-08-10 by killianstorm