mjbots/moteus #38: [Python] Not specified version for importlib_metadata causes build to fail for older moteus versions

bakowroc, 2022-10-12:
Hi,

we use `moteus` in our project but we override some of the `FdcanusbFactory` (and bunch of related classes) using monkeypatching due to need of the features which are not present in any `moteus` version. Because of  that we expect `moteus` to resolve always to the same version in order to avoid conflicts. Currently used version is `moteus==0.3.39`.

As of today we found out that `moteus` suddenly stopped working for us which happened to be connected with import from `importlib_metadata`, quick scan of `moteus` `setup.py` revealed an issue, which turned out to be that `moteus` does not specify version of the `importlib_metadata` hence each time a `moteus` is being built it enforces the newest version of `importlib_metadata`, which as for today is `5.0.0` (previously `4.11.3`).

We managed to use previous version of `importlib_metadata` by providing a specific version in our `pyproject.toml` dependencies (we use poetry) but nevertheless problem remains and this solution is not ideal, because it's gonna break as soon as we bump the moteus to the version that works with `importlib_metadata==5.0.0`.

Would be wonderful to have version of `importlib_metadata` specified in `setup.py`.

https://github.com/mjbots/moteus/blob/main/lib/python/setup.py.TPL

```
   install_requires = [
        'pyserial>=3.5',
        'python-can>=3.3',
        'pyelftools>=0.26',
        'importlib_metadata',
        'pywin32;platform_system=="Windows"',
    ]
```

bakowroc, 2022-10-12:
> ```
> 'importlib_metadata>=3.6',
> ```

Yes, that should work :)

jpieper, 2022-10-12:
Pushed in 0.3.47

https://pypi.org/project/moteus/0.3.47/

Look good?

bakowroc, 2022-10-12:
Sweet, many thanks!

closed 2022-10-12 by bakowroc