mjbots/moteus #46: Just reading data from controller

Andreasgdp, 2023-08-24:
When using either cpp og python library, I can't seem to figure out how to just read the data from the controller. 

# Use case
When developing a control system it is required to be able to know the states of the system at all times, i.e. also before any movement action has taken place.

# Current examples
In the examples given, the data is gotten after requesting a movement like:

```Python
state = await c.set_position(position=math.nan, query=True)

        # Print out everything.
        print(state)

        # Print out just the position register.
        print("Position:", state.values[moteus.Register.POSITION])
```

or 

```cpp
    cmd.position = std::numeric_limits<double>::quiet_NaN();
    cmd.velocity = 0.0;

    const auto maybe_result = controller.SetPosition(cmd);
    if (maybe_result) {
      const auto r = maybe_result->values;
      ::printf("%3d p/v/t=(%7.3f,%7.3f,%7.3f)  v/t/f=(%5.1f,%5.1f,%3d)   \r",
             static_cast<int>(r.mode),
             r.position,
             r.velocity,
             r.torque,
             r.voltage,
             r.temperature,
             r.fault);
      ::fflush(stdout);
    }
```

# Current alternative being used
I am primarily using CPP and have been using @sinaaghil's wrapper for the moteus driver, since he has implemented a way to read state, and not first request a position change.
Example:
https://github.com/sinaaghli/moteusapi/blob/main/example_internal/main_readstate.cpp

# If already possible
If it is already possible to just read the state using the libraries, it would be appreciated if an example of how this is done, was added to the libraries for ease of use and understanding.

jpieper, 2023-08-24:
In python do:

```
result = await c.query()
```

And in c++, do:

```
auto maybe_values = c.SetQuery();
```

Is that what you are looking for?

jpieper, 2023-08-24:
While there is not standalone documentation, the main python file and the main C++ header file are designed to be easy to read in order to discover all the available APIs:

https://github.com/mjbots/moteus/blob/035fe7915c2f9ddf4a25b15b0bc5fddc7e77f4a1/lib/python/moteus/moteus.py#L581
https://github.com/mjbots/moteus/blob/035fe7915c2f9ddf4a25b15b0bc5fddc7e77f4a1/lib/cpp/mjbots/moteus/moteus.h#L34

jpieper, 2023-08-24:
And finally, there are examples of using both APIs in the existing examples:

https://github.com/mjbots/moteus/blob/035fe7915c2f9ddf4a25b15b0bc5fddc7e77f4a1/lib/python/examples/gpio.py#L57
https://github.com/mjbots/moteus/blob/035fe7915c2f9ddf4a25b15b0bc5fddc7e77f4a1/lib/cpp/examples/nlateral_teleop.cc#L249

Andreasgdp, 2023-08-24:
Yes perfect `auto maybe_values = c.SetQuery();` is working flawlessly. For me it just wasn't intiutive that SetQuery() was responsible for reading data off of the controller.

jpieper, 2023-08-24:
For now I added a comment in 1bb210fe8c8a06f496c66952715e3afd692ad318

Other potential names for that function all have various problems, although I welcome votes for alternatives.

closed 2023-08-24 by jpieper