mjbots/moteus #47: ASM encoder bit shift when using I2C

matthieuvigne, 2023-09-09:
As you have noticed it your commit 6ae66ef on July 23rd, the AS5048B/AS5600 encoders have a 2-bit shift in the I2C registers: 0XFF holds only 6 LSB, while 0xFE hold the other bits.

Previous versions of the firmware (on which my fork was based) did not include this bit shift, so I corrected it to:
```c++
    status->value =
        (encoder_raw_data_[4] << 6) |
        (encoder_raw_data_[5]);
```

This way, I get the 14-bit raw value of the encoder (i.e. one turn goes from 0 to 16384) ; the upper 2 bits of the uint16_t are set to zero.

I initially planned on opening a PR for this bugfix, but then I noticed that on 6ae66ef you modified this to:

```c++
    status->value =
        (encoder_raw_data_[4] << 8) |
        (encoder_raw_data_[5] << 2);
 ```

This means that you chose the have the lowest 2 bits at 0 - in other words, this encoder goes from 0 to 65536 by increments of 4. Was this done on purpose, i.e. do you want to fake a 16-bit encoder when the hardware only has 14 bits ? Or is this a mistake ?

jpieper, 2023-09-09:
The intent was to have it use the 16 bit range.  That shouldn't be a problem, as long as the CPR is configured correctly.

matthieuvigne, 2023-09-09:
Ok makes sense thanks. To be honest I find it a bit weird to configure a cpr at 16 bits for a 14 bits encoder, but as you say it all works perfectly fine, so I guess in the end it's just a question of taste.

Closing this issue.

closed 2023-09-09 by matthieuvigne

jpieper, 2023-09-09:
Agreed it doesn't make the most sense... when I first implemented support I thought it reported a 16 bit value.  Now changing it would require adding upgrade logic in moteus_tool so as to not break systems on fw upgrade, which doesn't seem worth it.