mjbots/moteus #49: OSx Builds

mawildoer, 2023-11-04:
Hi Josh,

Great looking project I'm trying to get my teeth into!

I've been running into a few issues building this on OSx, but reading through mainly the Bazel configs, I see only a handful of places which we will likely need to append to to make it work.

I expect if I can figure out how to make this work on OSx, we'll likely be able to do the same for Linux as well.

I'm gonna keep at it and see how far I can get

## What I've found so far

### 1

Off the bat, executing even `bazel --help` from the workspace causes the following error:
```
(.venv) moteus % bazel --help
Traceback (most recent call last):
  File "/Users/mattwildoer/Projects/mjbots/moteus/tools/bazel", line 107, in <module>
    main()
  File "/Users/mattwildoer/Projects/mjbots/moteus/tools/bazel", line 102, in main
    os.execv(bazel_bin_loc, ['bazel', ] + sys.argv[1:])
OSError: [Errno 8] Exec format error
```

At surface level, it seems like `tools/bazel` achieves the same thing as [bazelisk](https://github.com/bazelbuild/bazelisk)?
Seems like a pretty old script and I'm wondering if it's still required?

### 2

I see Windows-only deps here: https://github.com/mjbots/moteus/blob/1f4a19554a2b99578cdf566ec5172ee7ddc8304f/WORKSPACE#L36-L49

I notice the place it's forked from has doubled in commits since it was forked to `mjbots` and appears to support Windows, Linux and OSx: https://github.com/grailbio/bazel-toolchain/blob/master/platforms/BUILD.bazel#L34

Is there anything critical we'd miss by moving back to it?

jpieper, 2023-11-05:
The only thing you can compile in this repo is the firmware and its tests.  Unfortunately, that is only supported on x64 Ubuntu.  Windows and OS X are not supported host platforms for firmware compilation.  If you need to compile custom firmware, you can use a VM image.  However, pre-compiled releases are available in github, so most people don't ever need to compile the firmware.

thestumbler, 2023-11-05:
I'll ditto Josh's reply. I went down that path last year and gave up. I concluded that if I knew enough about Bazel, and I *really* wanted to run it on a Mac, I could have eventually solved it -- maybe. But that didn't apply in my case and it was easy enough to build firmware on an Ubuntu system, either VM (like parallels) or a separate machine. 

I only did this briefly to experiment with some non-standard baud rates, but in the end I solved my CAN FD baud issues and used stock firmware.

mawildoer, 2023-11-05:
Oh, that's totally fine! I'm more than happy to use ubuntu in a docker container or the likes! I just thought it worked on Windows-only and wanted to fix that.

Did I perhaps miss some docs on "building firmware", or is the code the docs for how to build that?

jpieper, 2023-11-05:
https://github.com/mjbots/moteus/blob/main/docs/reference.md#flashing-and-building-firmware

mawildoer, 2023-11-05:
Sleeping on it, I came up with another approach which I hope can help others get off the ground quickly: https://github.com/mjbots/moteus/pull/50

closed 2024-08-01 by jpieper