mjbots/moteus #59: Moteus GUI Segfaults

kylevernyi, 2024-07-19:
I am trying to run the moteus_gui / tview tools to connect to my moteus-c1 for the first time. I am following the instructions of the getting started [video here](https://www.youtube.com/watch?v=6prAv9hbmeM&t=164s).

I am running Fedora 40. I ran the `pip3 install moteus-gui` command successfully. When I run `python3 -m moteus_gui.tview` command I get the following seg fault:

`[1]    8384 segmentation fault (core dumped)  python3 -m moteus_gui.tview`

I tried running gdb with `gdb --args python3 -m moteus_gui.tview`. I have attached the backtrace. [backtrace.txt](https://github.com/user-attachments/files/16315248/backtrace.txt) I put it into chatGPT and got the following which may or may not be correct:

> The stack trace indicates that the segmentation fault occurs when Python is trying to call the PyType_Ready function with a NULL type pointer. This suggests that there is a problem with the initialization of a Python type object, possibly due to an issue with the shiboken6 library.

I reinstalled the shiboken6 library but it had no effect. 

Thank you for the support.

jpieper, 2024-07-19:
Hi!  That is a known problem with pyside6.7.  The pip packages require a version older than 6.7, but it loos like Fedora 40 has pyside 6.7 as a system package which you may have installed through yum.   Are you familiar with using python virtual environments?  If so, as a workaround you could use one with site packages disabled.

```
python -m venv --no-site-packages venv
venv/bin/pip install moteus-gui
venv/bin/tview
```

jpieper, 2024-07-19:
For reference, the upstream bug we filed is: https://github.com/jupyter/qtconsole/issues/605

kylevernyi, 2024-07-19:
Wow thanks for the quick fix. I am running python3 so I had to change your command slightly to

`python -m venv venv`
`venv/bin/pip install moteus-gui`

Unfortunately PySide 6.6 was still installed in the venv for some reason. I did a manual update with 

`pip install PySide6 --upgrade`

Confirmed with 

`python -c "from PySide6 import __version__; print(__version__)"`

Got the output to be 6.7.2 which is what you mentioned needing. 

Then I ran `tview` 

And everything is working!

[image]

Thanks for the quick help. Looking forward to using these controllers. We are building a satellite simulator in my lab and are using these for the gyroscopes.