mjbots/moteus #62: Moteus R4.11 uses a linear incremental encoder to control the timing timeout of linear motor calibration

away12356, 2024-07-23:
@jpieper

jpieper, 2024-07-23:
moteus does not currently support linear motors at all.

Is this a feature request?

away12356, 2024-07-23:
Can it only be used for rotating motors？

jpieper, 2024-07-23:
Correct, only rotating motors.

The firmware and tools are open source, so you could implement support yourself.  If you did so, you would certainly be welcome to share any results back, but it isn't required.

away12356, 2024-07-25:
Hi Josh,
How to use the can send command? Display ERR unknown command on tview
[image]

away12356, 2024-07-25:
[image]

away12356, 2024-07-25:
[image]

away12356, 2024-07-25:
@jpieper

closed 2024-07-25 by away12356