mjbots/moteus #64: Hello, may I ask how to port the Moteus API to Qt?

away12356, 2024-08-01:


jpieper, 2024-08-01:
I'm unsure exactly what you are asking.  If the question is just how to use one of the moteus client side bindings from within a QT program, then there are several options.

First, you could use the python bindings along with the `asyncqt` library like tview does:

* https://github.com/mjbots/moteus/blob/main/utils/gui/moteus_gui/tview.py
* https://pypi.org/project/asyncqt/

Second, on linux or mac you could use the C++ bindings in either blocking or async mode.  One example (not developed or endorsed by mjbots) is here:

* https://github.com/georgevargas/GUI-APP-for-Moteus-Motor-Control

Does that answer your question?

away12356, 2024-08-01:
Is the minimum accuracy of Moteus position control 0.0036

jpieper, 2024-08-01:
The resolution of the onboard encoder is 14 bit, which works out to 0.000061 revolutions or 0.0219 degrees.  However that is only the resolution and not the accuracy.  In practice it has around 11-12 bits of precision depending upon your magnet mounting.  A rule of thumb is that position control needs about 16 counts of actual resolution of the encoder for typical loads.  That would result in a practical position control accuracy using the onboard encoder of around 1.4 degrees or 0.0039 revolutions.

The position control accuracy largely depends upon the encoder used.  If you have a more precise external encoder, then moteus can control position to an accuracy dependent upon that encoder using the above rules of thumb.

away12356, 2024-08-01:
What is the maximum resolution supported by Moteus?

jpieper, 2024-08-01:
The supported encoder with the current highest resolution is the 22 bit AksIM-2.  Internally, the position is tracked as a fixed point value with 32 bits of resolution per revolution as measured at the output.

away12356, 2024-08-01:
Do you recommend a high-resolution magnetic encoder

jpieper, 2024-08-01:
The encoder to use depends entirely on your application.  In many applications, the onboard encoder of moteus is sufficient.  Do you have any specific requirements you are targeting?

away12356, 2024-08-01:
The precision requirement for position control is relatively high, and it needs to be controlled very stably

away12356, 2024-08-01:
In practical applications, it has been found that the position control is not very stable when using onboard encoders

away12356, 2024-08-01:
The load is an optical instrument, which is hit on a mirror surface with a laser, causing significant shaking of the light spot

away12356, 2024-08-01:
Resolution 0.0001 degrees

jpieper, 2024-08-01:
None of the absolute encoders supported by moteus currently will be able achieve that resolution.  The 22 bit AksIM-2 has only around 0.0001 degree of resolution (I don't know its effective accuracy, but it is almost certainly worse), and as mentioned above you will need at least 16 counts of actual accuracy to achieve position control.

If you found a quadrature encoder with appropriate resolution, you might be able to use it on a moteus-n1 or moteus-c1 with the hardware quadrature inputs, however experimentation would be required.

Unfortunately, I don't have any specific encoder to recommend which can achieve your requested accuracy.

away12356, 2024-08-01:
Can incremental encoders be used

away12356, 2024-08-01:
What is the maximum number of bits supported by an incremental encoder?

closed 2024-08-06 by away12356