mjbots/moteus #66: runtime error

Chetanyadav03, 2024-08-03:
while running the moteus BLDC motor with the r4.11 controllor(moteus), the calibration process showed an error for runtime, saying the firmware doesnot support the direction of the Motor. The exact error is:
RuntimeError: Requested motor direction not possible with current firmware version
What can be done to remove this error, it worked few hours earlier, but stopped working as i updated my windows

jpieper, 2024-08-03:
You can additionally pass `--cal-invert` to the calibration command or upgrade the firmware of your moteus.

Chetanyadav03, 2024-08-03:
Can u please guide on how to upgrade the firmware for the moteus controller

jpieper, 2024-08-03:
https://github.com/mjbots/moteus/blob/main/docs/reference.md#flashing-over-can

jpieper, 2024-08-04:
Was that enough to get your firmware updated?

Note, that passing `--cal-invert` is an equally fine option.

Chetanyadav03, 2024-08-04:
Yes it worked... Thanks a lot for your help... Before this I wasted like 3
hrs trying to rectify it but after your help it took about 10 minutes ...
👍🏼

jpieper, 2024-08-04:
Great!

closed 2024-08-04 by jpieper