mjbots/moteus #68: Can Moteus work at -45 ° C-100 ° C

away12356, 2024-08-15:


jpieper, 2024-08-15:
All components on the board are rated at least -40C to 85C, however the board does not have rated temperature bounds. Any requirements would need to be evaluated in a specific application.

thestumbler, 2024-08-15:
One issue with plastic packages ICs at extremely cold temperatures is that the coefficient of expansion is different between the plastic material and the internals of the chip. As a result one main failure mode is that the internal lead wires can pull apart. At least this is what I was told as a young engineer. I’m not 100% if that logic applies to BGAs or not, but I would be hesitant to tempt fate. 

This is one reason military and other high reliability projects use ceramic packages, because those are able to perform down to -55° (I don’t know what makes those fail below that point). That’s why many ICs used to be available in either plastic or ceramic packages, but that’s not a given anymore. 

And Josh is correct in that having suitably rated ICs and other components is necessary but not sufficient to rate a board over a certain temperature range. Other factors like the PCB design and assembly, heat sinking, etc., are important as well. You’d probably want to conformal coat a board that goes down to -45 also, I’m thinking about moisture getting under a chip and freezing, for one example.

jpieper, 2024-08-22:
I'm going to consider this concluded given the lack of response.

closed 2024-08-22 by jpieper