mjbots/moteus #70: Motor calibrates but will not move.

robo-todd, 2024-08-22:
We have had great luck using the Moteus r4.11 controllers thus far with your default motor. We're moving on in our project to using a wide variety of different motors. Currently we're bringing up various FlipSky motors like a 6384/190kv motor.

The calibration procedure generates a calibration for the motor and I can even use the "d pwm" command to cause it to move, but position mode does nothing. It does not even seem to *try* to move the motor at all whereas your default motor just worked.  Any thoughts on how to get things moving again. These motors have a lot of cogging torque for sure. Suggestions?

I'm attaching
[flipsky.txt](https://github.com/user-attachments/files/16711453/flipsky.txt)
  the output of the default calibration which seems to work / behave pretty much the same on your default motors as this motor.

jpieper, 2024-08-22:
If the motor does not move you need to look at the reported mode and fault.  In tview, they are the first two entries in the `servo_stats` tab.

Fault decodings can be found here: https://github.com/mjbots/moteus/blob/main/docs/reference.md#0x00f---fault-code

jpieper, 2024-08-22:
Oh, and if you're integrating a bare controller for the first time, you might want to look through the points in the getting started guide: https://github.com/mjbots/moteus/blob/main/docs/getting_started.md including the configuration options you will likely want to ensure are usable.

robo-todd, 2024-08-22:
Thank you. So I'm getting fault 39, but I thought that I set the position limits and I did issue a stop command to reset mode, etc. It was strange because even "d brake" was not working.

Before you had answered I tried a few things like forcing the KV to the label on the motor, and then it seemed to work. I'm currently running an automatic config (not forced) and it is working as well, but on startup it had the fault 39. I'm going to try restarting a couple times (power cycle) just to see what's up.

Thanks for the quick response.

I see a lot of parameters for cogging_dq_comp and these are very high cogging torque motors. They are all zero after my default calibration. Is there a way to use those to smooth things out that I'm missing somewhere? I only see selecting higher or lower cal_bw_hz for this?

robo-todd, 2024-08-22:
OK... I forgot to save position limits in the calibration. It comes up with fault 39 basically every time because of that. Thanks for your answers. 

Question on if I can optimize cogging torque comp still remains. I don't really see much in the reference or getting started on it (or I'm missing it).

jpieper, 2024-08-22:
If you are in a fault state, then the only command which will work is: `d stop`

You may have set the limits, but not issued a `conf write` to cause them to be stored to persistent storage?  If you restart tview, it will re-update all configuration parameters so you can see what is set now.

Cogging compensation is experimental.  I'd recommend getting things to work as well as you can before messing with it: https://docs.google.com/document/d/1M4O-LVki03qonNTkydv1N8gRteGIZk3ZyHd-mBpxL-U/edit#heading=h.x1qddq7g5cts

jpieper, 2024-08-22:
Checking in to see if you're all set here?  If so, I can close this ticket out.

robo-todd, 2024-08-22:
Yes. All good here. Was user error on saving position limits.

jpieper, 2024-08-22:
Great, thanks!

closed 2024-08-22 by jpieper