mjbots/moteus #74: How to maximize the output torque when the motor is stuck

away12356, 2024-11-11:


jpieper, 2024-11-11:
Unless you provide a command specific torque limit, are operating near the thermal limit, or have decreased the configurable maximum power or maximum current limit, moteus already applies the maximum torque possible.

An invalid calibration, caused by a poorly secured magnet or magnet that has exceeded its operating temperature, can also limit torque.  You could confirm that was the problem if calibrating the motor resolved the issue.

Can you provide any more context?

away12356, 2024-11-11:
When stall, the maximum output torque of the motor is 1.5N · m and the current is 6A. Is there any way to increase the output torque

jpieper, 2024-11-11:
What motor are you using, what are its specifications, and what input supply voltage are you using?

jpieper, 2024-11-11:
And where are you measuring current?

away12356, 2024-11-11:
PMSM, input voltage 24V

away12356, 2024-11-11:
Measure the current at the input end

jpieper, 2024-11-11:
What is the Kv, and winding resistance of your motor?

away12356, 2024-11-11:
Kv: 16.03763626929022
winding resistance: 2ohm

jpieper, 2024-11-11:
The Kv value looks like it was measured by moteus.  Was the winding resistance measured by moteus, or is it the nameplate number?

jpieper, 2024-11-11:
Thanks!  I would expect moteus to be able to apply approximately 11A of phase current to that motor with a 24V supply, which should result in about 6Nm of torque, assuming the motor does not become magnetically saturated.  I will say that your reported Kv rating is abnormally low for motors I have seen.

Is your motor an off the shelf motor?  Can you provide a link?  Does it have a nameplate peak or continuous torque rating?

Can you share a screenshot of tview when maximum torque is being generated?  I would like `servo_stats` expanded, the bottom command window included in the screenshot, and `servo_stats.torque_Nm` plotted on the left axis and `servo_stats.q_A` plotted on the right.

away12356, 2024-11-11:
[image]

jpieper, 2024-11-11:
In that plot, moteus reported that it generated a peak of 4Nm of torque.  Have you independently measured the torque and believe it to be only 1.5Nm instead of 4?

Also, from that plot it isn't clear what, if any were limits were reached during that process.  Can you share a screenshot of tview showing the telemetry tab with `servo_stats` while the maximum torque is being produced?

Finally, I'll repeat my question of if the motor you are using is off the shelf.  If so, can you provide a link?  Does it have a nameplate continuous or peak torque rating?

away12356, 2024-11-11:
This motor was designed by ourselves

away12356, 2024-11-11:
[image]

jpieper, 2024-11-11:
Thanks, however I need to see the contents of `servo_stats` *while* the motor is outputting the peak torque, not after.

jpieper, 2024-11-11:
Actually, while I'm asking for more things, instead of plotting, `servo_stats.q_A`, can you plot `servo_stats.position` on the right axis during the test?

away12356, 2024-11-11:
[image]

jpieper, 2024-11-11:
Thank you.  In this plot moteus claims it is drawing about 340W of power from your supply and the voltage of your supply has sagged to 22V, possibly from resistive losses between your PSU and the controller.

It looks like you are at, or near several different limits.

I believe the limit you are hitting right now is likely the phase modulation limit, which is derived from your supply voltage and phase resistance.

However, even you increased your supply voltage, you are very close to the electrical power limit of the moteus-r4, so you would not be able to increase the current or torque very much more, probably only by about 15% more before you hit the electrical power limit of moteus-r4.

moteus-n1 has a higher electrical power limit of 1200W, although the default in config is still 450.  

Thus, if you want to get more torque out of this motor with a moteus controller, I would recommend increasing your supply voltage to 36 or 48V and use a moteus-n1 with `servo.max_power_W=1200`.  You are likely to still be limited by the maximum power rating of the moteus controller, but it will at least be a higher limit.

Note, that this maximum power limit will become more constraining if you are moving at non-zero speed, and if you need to achieve this maximum torque at non-zero speed it may not be practical to use a moteus controller with this motor.

away12356, 2024-11-11:
Can I increase the voltage to 44V

jpieper, 2024-11-11:
You can increase the voltage from 24V.  For moteus-r4, the 44V rating is the absolute maximum.  You will likely need to leave some margin to handle transients that result from regenerative braking at a minimum.  See: https://github.com/mjbots/moteus/blob/main/docs/reference.md#regenerative-braking-safety

As mentioned though, with the moteus-r4 and this motor, I expect you will hit the maximum power limit with a voltage of even 36V, so going higher than that with a moteus-r4 adds no more performance but increases the risk of damage.

away12356, 2024-11-12:
[image]

away12356, 2024-11-12:
Why can't PWM reach all high or low values during stalling

jpieper, 2024-11-12:
Currently moteus has an approximately 90% maximum duty cycle, partially due to the need to sense current and partially due to waveform generation limitations.

away12356, 2024-11-13:
Does tview have the function of recording data

jpieper, 2024-11-13:
tview does not have a data recording feature.  One easy option is to make a python script that performs your action, then do something like this example to write the data to a CSV file: https://github.com/mjbots/moteus/blob/main/lib/python/examples/csv_output.py

jpieper, 2024-11-14:
The position achievable by moteus will depend upon the encoder used.  See this previous issue where you asked a similar question: https://github.com/mjbots/moteus/issues/64

away12356, 2024-11-15:
What is the maximum frequency that CANFD supports for sending data? 
results = await c.set_position(position=pos, kp_scale=5.0,kd_scale=1.0, accel_limit=1.0, query=True)
This function will block 2ms, can it not be blocked

jpieper, 2024-11-15:
You can read about CAN-FD update frequency here: https://jpieper.com/2024/05/16/optimizing-moteus-command-rate/

The python library uses the python asyncio primitives.  You can use any of the facilities available in that library to manage concurrency: https://docs.python.org/3/library/asyncio.html

closed 2024-11-18 by away12356