mjbots/moteus #75: T-Motor MN5008 Calibration

robo-todd, 2024-11-15:
Hello. We've been having great luck with your R4.11 controller on a huge variety of motors and we've come up against some T-Motor MN5008's KV170's in our design and they simply will not calibrate, even with forcing the KV and trying to set the number of poles, etc.  It gets through the encoder calibration, etc. but then fails on KV and even forcing KV the resulting setup will not spin the motor.

Any suggestions on how we can resolve this? First one of similar motor that didn't "just work" but we need them for our design, so I'm hoping there is a way around it.

jpieper, 2024-11-15:
Have you worked through the items in this post?

https://jpieper.com/2024/05/23/debugging-moteus-calibration-failures/

robo-todd, 2024-11-15:
The motors are completely free to spin when we're doing this but I've not increased the power. Soldering should be solid and this is common to about 10 motor/amp setups so I don't think we have soldering issue on all 10 (about 10 other setups we have soldered in the same way with different motors all worked just fine basically instantly). There are no encoder error complaints from cal and the metric gets down to about 20-22.

I can try higher power, but it does spin completely around/back during the encoder calibration part... but on Kv it just never makes the motor(s) move.

jpieper, 2024-11-15:
As mentioned in that document, encoder problems will often not report that there is an encoder problem.  There are 95% odds that your problem is with the sense magnet.  Can you read through that post and verify all the sense magnet things it mentions?

robo-todd, 2024-11-15:
Yes. Turns out it is a sense magnet problem. Our mechanical folks thought they had used the same parameters, but we found an issue. We're making a test jig for the setup to double check with our mounting hardware out of the mechanical loop. This is likely mechanical. I'll try to close the post with an update today/tomorrow on what we found out.

robo-todd, 2024-11-16:
Following up. We had misaligned the mount (laterally offset), The MN5008 calibrated fine after we checked it on a separate jig. Thanks for the quick responses!

closed 2024-11-16 by robo-todd