mjbots/moteus #76: Can't the resolution of Moteus using incremental encoders exceed 200000

away12356, 2024-11-26:


jpieper, 2024-11-26:
Are you observing an unexpected behavior?  If so, please describe it.

away12356, 2024-11-26:
Manually rotate the motor quickly, with no change in CPR and a large amount of error data

away12356, 2024-11-26:
R11, I am using software orthogonal. How can I connect hardware orthogonal

jpieper, 2024-11-26:
With software quadrature, you are limited to roughly 100000 counts per second of motion, no matter what the resolution is. 

Unfortunately, hardware quadrature is not supported on moteus-r4, only on moteus-c1 and moteus-n1.  With those boards and hardware quadrature, the limit is higher at roughly 10000000 counts per second.

away12356, 2024-11-26:
What is the difference between hardware orthogonality and software orthogonality

jpieper, 2024-11-26:
Software quadrature is implemented as a software interrupt, hardware quadrature uses a dedicated hardware decoder in the microcontroller.

away12356, 2024-11-26:
Is hardware orthogonal wiring the same as software orthogonal wiring

away12356, 2024-11-26:
硬件正交布线与软件正交布线相同吗？

away12356, 2024-11-26:
Hardware quadrature wiring is the same as software quadrature?

closed 2024-11-28 by away12356