mjbots/moteus #80: Support for secondary encoder

gupta-divy, 2025-01-10:
I recemtly purchased a moteus c1 motor controller and MA600 breakout board. I am using the onboard encoder for motor as in the devKit and was hoping to use the aux2 with MA600 breakout as secondary encoder for another shaft rotation in my system. Is it possible with current firmware? Can I directly read the aux2 SPI for encoder reading. Currently, the GUI shows spi_pin_error for aux2 SPI with the MA600 encoder mounted. 

Any suggestions to make the setup work in this config?

jpieper, 2025-01-11:
First, you can often get faster support in the mjbots Discord: https://discord.gg/W4hUpBb

Second, `spi_pin_error` means you have not configured pins for `aux2` which are compatible with a SPI sensor.  You should set

```
aux2.pins.0.mode 1  # spi
aux2.pins.1.mode 1  # spi
aux2.pins.2 mode 1  # spi
aux2.pins.3.mode 2  # spi_cs
```

Note, you will also need to `motor_position` if you want to use the SPI encoder for anything.

You can read more about this in the reference manual: https://github.com/mjbots/moteus/blob/main/docs/reference.md#encoder-configuration and in this video: https://www.youtube.com/watch?v=a3UA6yzzgkE

gupta-divy, 2025-01-13:
Thanks, setting the pin configuration worked!

closed 2025-01-13 by gupta-divy