mjbots/moteus #83: Error code 33

acastles91, 2025-02-15:
My c1 driver was working this morning but now it's not moving my motor and servo_stats.mode is fault 1 and the fault is code 33. I have another c1 driver attached to the same kind of motor, with the same configuration and the same power supply and that one works without a problem, what can I do?

jpieper, 2025-02-15:
Fault 33 means the gate driver has reported a fault.

https://github.com/mjbots/moteus/blob/main/docs/reference.md#0x00f---fault-code

In the best case, you have a transient under voltage caused by drawing more current than your supply can provide.  If the fault 33 is persistent, then the board is likely damaged electrically.

What power supply and motor are you using?  What was the system doing the last time it was working?

acastles91, 2025-02-15:
Thanks for the help! The power supply is the 24V provided with the development kit and the motor is a [57BLF01](https://www.act-motor.com/brushless-dc-motor-57blf-product/). The last time it was working the motor was doing this command:

d pos -5000 0 0.9 v30.0 a10.0

I was testing how fast I could get to move a sculpture that I'm designing.

jpieper, 2025-02-15:
Unfortunately, I don't have much help at the moment and am about to go afk for 8 hours, but I don't see any particular problem with that motor.

While you were running that command did the motor do anything unusual?

acastles91, 2025-02-15:
Thank you for the fast reply. No, the motor didn't do anything strange, it performed the command but then it didn't move again. Is it possible that the motor is over sized for the driver and damaged it? Because I have a similar setup, the sculpture I'm designing needs to motors and I wouldn't want to damage the other driver

jpieper, 2025-02-15:
There is no problem with the size of the motor.  If you haven't yet, you should configure flux braking per the deployment guidelines: 

https://github.com/mjbots/moteus/blob/main/docs/reference.md#regenerative-braking-safety

That said, we would like to make it such that you can't damage the device even if flux braking is not configured properly.  I would like to try and reproduce your problem and can get you a replacement driver, but not until after our vacation, so the 24th. To get that started, can you send an email to info@mjbots.com with your order number?

In the meantime, do you have an estimate for how much inertia or mass you have on the motor?

jpieper, 2025-02-24:
We have tackled this via email.

closed 2025-02-24 by jpieper

jtoobias, 2025-10-30:
what was the solution? I am getting the same issue

jpieper, 2025-10-30:
There are many potential causes of fault 33.  Rather than re-opening a stale issue, would you mind making a new one or sending a request by email or Discord?

You probably should work through the troubleshooting guide for that first as well: https://mjbots.github.io/moteus/troubleshooting/gate-driver-faults/