mjbots/moteus #85: Abnormal calibration

away12356, 2025-04-11:
Calibrating - 
Initial metric: 1708.3828481377807
optimizing - metric=25.34327        
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/bmr/.local/share/pipx/venvs/moteus/lib/python3.12/site-packages/moteus/moteus_tool.py", line 2288, in <module>
    main()
  File "/home/bmr/.local/share/pipx/venvs/moteus/lib/python3.12/site-packages/moteus/moteus_tool.py", line 2284, in main
    asyncio.run(async_main())
  File "/usr/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/bmr/.local/share/pipx/venvs/moteus/lib/python3.12/site-packages/moteus/moteus_tool.py", line 2280, in async_main
    await runner.start()
  File "/home/bmr/.local/share/pipx/venvs/moteus/lib/python3.12/site-packages/moteus/moteus_tool.py", line 2069, in start
    await self.run_action(target)
  File "/home/bmr/.local/share/pipx/venvs/moteus/lib/python3.12/site-packages/moteus/moteus_tool.py", line 2142, in run_action
    await stream.do_calibrate()
  File "/home/bmr/.local/share/pipx/venvs/moteus/lib/python3.12/site-packages/moteus/moteus_tool.py", line 1129, in do_calibrate
    cal_result = await self.calibrate_encoder_mapping(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bmr/.local/share/pipx/venvs/moteus/lib/python3.12/site-packages/moteus/moteus_tool.py", line 1261, in calibrate_encoder_mapping
    return await self.calibrate_encoder_mapping_absolute(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bmr/.local/share/pipx/venvs/moteus/lib/python3.12/site-packages/moteus/moteus_tool.py", line 1436, in calibrate_encoder_mapping_absolute
    raise RuntimeError(f"Error(s) calibrating: {cal_result.errors}")
RuntimeError: Error(s) calibrating: ['encoder not an integral multiple of phase, 0.37255632098305735 > 0.1']

jpieper, 2025-04-11:
See: https://jpieper.com/2024/05/23/debugging-moteus-calibration-failures/

away12356, 2025-04-12:
The optical encoder I use

jpieper, 2025-04-12:
Your most recent comment appears truncated.

closed 2025-04-22 by away12356