mjbots/moteus #86: 在构建过程中报告错误

away12356, 2025-04-23:
ERROR: no such package '@llvm_toolchain//': Unknown LLVM release: clang+llvm-10.0.0-x86_64-linux-gnu-ubuntu-16.04.tar.xz

jpieper, 2025-04-23:
What host operating system and version are you using?

away12356, 2025-04-24:
ubuntu22.04，moteus-20240520

jpieper, 2025-04-24:
Only x64 desktop with 20.04, 22.04, and 24.04 is supported.  My guess is that you are using a different architecture.

away12356, 2025-04-24:
x86_64

away12356, 2025-04-24:
amd64

jpieper, 2025-04-24:
Is this your first time attempting to compile?  Have you had success elsewhere?

jpieper, 2025-04-24:
That error I would expect if you were trying to build on 24.04 with that older version of the moteus source.  I would expect 22.04 to work though.  In any event, can you build the most recent version?

jpieper, 2025-04-24:
The first release version that supports building on 24.04 is 2024-010-29.  Support was added in this commit: https://github.com/mjbots/moteus/commit/ee4f6ac9c6391523876836fcf6f276411b7f3670

away12356, 2025-04-24:
I am using Ubuntu24.04 to build the latest version of motues. Error message:< builtin>: BazelWorkspaceStatusAction stable-status.txt failed: Failed to determine workspace status: Process exited with status 127
/bin/sh:  1:  tools/workspace_status.sh: not found， Why was this file successfully built when it was stored in the tools directory

jpieper, 2025-04-24:
Are you building from a git checkout?  That is the only supported method of building.

away12356, 2025-04-24:
ERROR: <builtin>: BazelWorkspaceStatusAction stable-status.txt failed: Failed to determine workspace status: Process exited with status 127
/bin/sh: 1: tools/workspace_status.sh: not found
Analyzing: target //:target (2 packages loaded, 0 targets configured)

FATAL: bazel crashed due to an internal error. Printing stack trace:
java.lang.UnsupportedOperationException: /mnt/hgfs/pro/moteus/bazel-bin (??????)
	at com.google.devtools.build.lib.unix.NativePosixFiles.symlink(Native Method)
	at com.google.devtools.build.lib.unix.UnixFileSystem.createSymbolicLink(UnixFileSystem.java:364)
	at com.google.devtools.build.lib.vfs.Path.createSymbolicLink(Path.java:483)
	at com.google.devtools.build.lib.vfs.FileSystemUtils.ensureSymbolicLink(FileSystemUtils.java:341)
	at com.google.devtools.build.lib.vfs.FileSystemUtils.ensureSymbolicLink(FileSystemUtils.java:292)
	at com.google.devtools.build.lib.buildtool.OutputDirectoryLinksUtils.createLink(OutputDirectoryLinksUtils.java:277)
	at com.google.devtools.build.lib.buildtool.OutputDirectoryLinksUtils.createOutputDirectoryLinks(OutputDirectoryLinksUtils.java:141)
	at com.google.devtools.build.lib.buildtool.ExecutionTool.createConvenienceSymlinks(ExecutionTool.java:814)
	at com.google.devtools.build.lib.buildtool.ExecutionTool.handleConvenienceSymlinks(ExecutionTool.java:745)
	at com.google.devtools.build.lib.buildtool.BuildTool.buildTargetsWithMergedAnalysisExecution(BuildTool.java:351)
	at com.google.devtools.build.lib.buildtool.BuildTool.buildTargets(BuildTool.java:173)
	at com.google.devtools.build.lib.buildtool.BuildTool.processRequest(BuildTool.java:510)
	at com.google.devtools.build.lib.buildtool.BuildTool.processRequest(BuildTool.java:478)
	at com.google.devtools.build.lib.runtime.commands.TestCommand.doTest(TestCommand.java:164)
	at com.google.devtools.build.lib.runtime.commands.TestCommand.exec(TestCommand.java:117)
	at com.google.devtools.build.lib.runtime.BlazeCommandDispatcher.execExclusively(BlazeCommandDispatcher.java:664)
	at com.google.devtools.build.lib.runtime.BlazeCommandDispatcher.exec(BlazeCommandDispatcher.java:244)
	at com.google.devtools.build.lib.server.GrpcServerImpl.executeCommand(GrpcServerImpl.java:573)
	at com.google.devtools.build.lib.server.GrpcServerImpl.lambda$run$1(GrpcServerImpl.java:641)
	at io.grpc.Context$1.run(Context.java:566)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)

away12356, 2025-04-24:
INFO: Analyzed target //:target (62 packages loaded, 22097 targets configured).
INFO: Found 1 target and 0 test targets...
Target //:target up-to-date:
  bazel-bin/fw/moteus.bin
  bazel-bin/fw/moteus.08000000.bin
  bazel-bin/fw/moteus.0800c000.bin
  bazel-bin/fw/moteus.08010000.bin
  bazel-bin/fw/can_bootloader.bin
INFO: Elapsed time: 900.149s, Critical Path: 14.27s
INFO: 439 processes: 3 internal, 436 processwrapper-sandbox.
INFO: Build completed successfully, 439 total actions
ERROR: No test targets were found, yet testing was requested

jpieper, 2025-04-24:
That looks like it was successful, you should be good to go.

away12356, 2025-04-24:
Thank you very much for your help

jpieper, 2025-04-24:
What boards do you have and which are produced by mjbots vs self produced?

away12356, 2025-04-24:
Can the latest version of mjbots' R4.11 program be used

jpieper, 2025-04-24:
Yes

away12356, 2025-04-24:
How to write UUID when downloading the program to a new board?

jpieper, 2025-04-24:
Unfortunately, mjbots only offers no-cost support to organizations that are exclusively using mjbots produced boards.  We do offer paid support options.  Send an email to info@mjbots.com for more information.

closed 2025-04-24 by jpieper