mjbots/moteus #87: I2C AS5600 Encoder and Digital Inputs

robo-todd, 2025-05-13:
We're working with the R4 and C1 controller boards and attempting to use AUX ports for connection to digital inputs (limit switches) and absolute encoders (AS5600 boards on I2C) and when we configure these things we get nothing back and see no activity.

Perhaps we're reading the pin mappings wrong for documentation, but we have tried a lot of permutations and see no activity on the pins and setting them to digital inputs for limits also never seem to get read (status panel of t-view for the port).

Any suggestions? We're getting nowhere and no configuration for aux2/aux1 seem to have any effect or result in any I2C activity or gpio input state being read.

jpieper, 2025-05-13:
For the r4 I2C, you need to use pins 0 and 1, per the "I2C" disambiguator section of: https://jpieper.com/2022/06/30/flexible-i-o-worked-examples/

```
conf set aux2.pins.0.mode 13        # i2c
conf set aux2.pins.1.mode 13        # i2c
```

The moteus-r4 ABS pins have hard-wired 2.2k pullup resistors installed on the two pins, the pin specific pullups will have no effect for it.

Once you have set the correct pins, can you share a screenshot of your config and telemetry for `aux2` in tview?

Let's look at the digital inputs after you have I2C working.

jpieper, 2025-05-13:
You can also see the appropriate pin numberings to use in the "Aux" column of the pin table in the reference documentation: https://github.com/mjbots/moteus/blob/main/docs/reference.md#aux2--abs

[image]

robo-todd, 2025-05-19:
OK. Circling back to this.  Probably user error on our part, but we had to actively go in and disable the SPI setting and then our other settings worked just fine.  Maybe the documentation could make this default a little clearer? I would have thought that it would come out of the box in a kind of "everything disabled" mode for the aux ports. So that is what was throwing us off.

Once we disabled SPI things worked great. Thank you!

closed 2025-05-19 by robo-todd

jpieper, 2025-05-19:
I'm glad you got things working.  The aux port SPI being configured by default is actually fixed in the most recent firmware.  In any event, the key thing is to look in the telemetry under aux1/aux2 for errors, which will tell you what is amiss.