mjbots/moteus #89: numpy 2.y support

g0hl1n, 2025-05-21:
Hi Josh,

first and foremost: thanks for you work! 👍  Moteus and the tooling around is great! 👏

I have a question regarding the required numpy version of the python code (`<2` currently). Are there any plans to update to numpy 2.y?

In your [commit messsage from 2024-07-03](https://github.com/mjbots/moteus/commit/ac9173a267b02df0ea202a7a787640d77d713cf2) you mentioned:
> numpy 2 is now out, but other things we depend on don't work with it.
> For now, require <2.

Do you remember what "other things we depend on" were back then? Maybe they are "numpy 2 ready" already 😀?

Speaking of numpy dependencies: Is there a reason `lib/python` (and therefore https://pypi.org/project/moteus/) depends on numpy at all? AFAICT numpy isn't used anywhere in the library part, is it? 🤔 

Thanks again for your work and time!

regards;g0hl1n

jpieper, 2025-05-21:
It depends on scipy, which depends on numpy. The explicit dependency listing is merely to specify the version.

It may very well work now. Want to check across 22.04, 24.04, Windows, and rpi os? :)

If not, I'll get to it in a week or two.

g0hl1n, 2025-05-21:
Thanks for your fast response!

> It depends on scipy, which depends on numpy. The explicit dependency listing is merely to specify the version.

Ok. That makes sense. Thanks for the clarification.

> It may very well work now. Want to check across 22.04, 24.04, Windows, and rpi os? :)

Sure. I can check on Ubuntu and latest Raspbian tomorrow.

But what's that "Windows" thing? 😅 😉
(Tbh, I don't have access to any Windows machine).

> If not, I'll get to it in a week or two.

g0hl1n, 2025-06-12:
Thank you @jpieper and sorry for not providing feedback in a timely manner.