mjbots/moteus #92: CAN-FD frames aren't properly padded

jbtheou, 2025-06-26:
I noticed an issue where when the format is adjusted on the fly, the behavior is unexpected (stall data is still sent).

Looking at the candump, noticed that the userspace is sending non-valid DLC

```
  can1  00008001  [22]  01 00 0A 0D 20 CD CC 4C 3F 0D 29 00 00 C0 40 11 00 11 0B 19 81 02
  can1       100  [16]  21 00 0A 21 0B 01 29 81 02 0B 01 00 00 50 50 50
  can1  00008001  [19]  01 00 0A 0D 20 CD CC 4C BF 0D 29 00 00 C0 40 11 00 11 0B
  can1       100  [06]  21 00 0A 21 0B 01
```

Looking at the code, I am not seeing any auto-padding logic for the requests. Is it expected? Am I missing the logic somewhere?

Thanks a lot

jbtheou, 2025-06-26:
Regarding the actual stall data issue, with a bit more data requested, if I stop requesting 0x101 (FW version), its still sent
```
  can1  00008001  [24]  01 00 0A 0D 20 CD CC 4C 3F 0D 29 00 00 C0 40 11 00 11 0B 11 0F 19 81 02
  can1       100  [16]  21 00 0A 21 0B 01 21 0F 00 29 81 02 0B 01 00 00
  can1  00008001  [21]  01 00 0A 0D 20 CD CC 4C BF 0D 29 00 00 C0 40 11 00 11 0B 11 0F
  can1       100  [16]  21 00 0A 21 0B 01 21 0F 00 29 81 02 0B 01 00 00
```
But I believe that due to the invalid DLC, the FW is reusing some stall data, that's why it's happening. So the DLC padding seems to be the main issue

jbtheou, 2025-06-26:
@jpieper Please let me know if you need additional information. Working on a fix on my end in //

jpieper, 2025-06-26:
What software and hardware are you using to send commands?

jpieper, 2025-06-26:
I mean, the wire format for CAN-FD doesn't even have a representation for a DLC of 19 bytes, so by definition moteus did not receive a frame with that length.

jpieper, 2025-06-26:
For reference, the python library rounds up to the nearest DLC here in the code:

https://github.com/mjbots/moteus/blob/d9a1b3505c91308036ffd905c0cecaa429a5d7e6/lib/python/moteus/fdcanusb.py#L199
https://github.com/mjbots/moteus/blob/d9a1b3505c91308036ffd905c0cecaa429a5d7e6/lib/python/moteus/pythoncan.py#L114

And the C++ library does so here:

https://github.com/mjbots/moteus/blob/d9a1b3505c91308036ffd905c0cecaa429a5d7e6/lib/cpp/mjbots/moteus/moteus_transport.h#L669

Although it looks like the C++ socketcan library does not currently round up.  Are you using the C++ client library?

(also, the firmware version register is largely meaningless and not worth requesting, as it is actually the "ABI version" not the firmware version)

jbtheou, 2025-06-26:
Exactly, and I think that's why the FW is catching some outdated data when parsing the memory.  Just using a customized example

```
// Copyright 2025 mjbots Robotic Systems, LLC.  info@mjbots.com
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/// @file
///
/// This example shows how to use Controller::SetPositionWaitComplete.

#include <signal.h>
#include <unistd.h>

#include <iostream>

#include "moteus.h"

void clean_routine(int s)
{
  using namespace mjbots;

  ::printf("\nCaught signal %d\n", s);
  moteus::Controller::Options options;
  options.id = 1;

  moteus::Controller controller(options);

  // Command a stop to the controller in order to clear any faults.
  controller.SetStop();

  exit(1);
}

int main(int argc, char** argv) {
  using namespace mjbots;

  /* Not a part of the header */
  constexpr uint16_t kFirmwareVersion = 0x101;

  /* Catch CTRL + C */
  signal(SIGINT, clean_routine);

  moteus::Controller::DefaultArgProcess(argc, argv);

  moteus::Controller::Options options;
  options.id = 1;

  // Only position and velocity are sent by default when sending
  // position mode commands.  If you want to send any other fields you
  // need to specify their resolution.  That can either be done here
  // in the `Controller::Options` structure, or on a per-command basis
  // using the format override argument.
  options.position_format.accel_limit = moteus::kFloat;

  /* We are not using velocity, we can ignore it */
  options.position_format.velocity = moteus::kIgnore;

  moteus::Controller controller(options);

  // When using the diagnostic protocol, it is important to know that
  // applications like tview may leave the controller "spewing" on the
  // diagnostic channel, i.e. sending unsolicited data.  In order stop
  // that, a client needs to issue a "tel stop" command, and then
  // flush all data that is present.  After that point, normal
  // commands can be issued.
  controller.DiagnosticWrite("tel stop\n");
  controller.DiagnosticFlush();

  // DiagnosticCommand will always return the result.  The `conf get`
  // diagnostic command is unique in that it replies with a single
  // line and no final "OK", so we use the `kExpectSingleLine` option.
  auto result = controller.DiagnosticCommand(
      "conf get servopos.position_max", moteus::Controller::kExpectSingleLine);
  if (result == "") {
    ::printf("Exiting because unable to get diag info (max).\n");
    return 1;
  }
  const auto pos_max = std::stod(result);

  result = controller.DiagnosticCommand(
      "conf get servopos.position_min", moteus::Controller::kExpectSingleLine);
  if (result == "") {
    ::printf("Exiting because unable to get diag info (min).\n");
    return 1;
  }
  const auto pos_min = std::stod(result);

  ::printf("Expected range: [%f,%f]\n", pos_min, pos_max);

  // Command a stop to the controller in order to clear any faults.
  controller.SetStop();

  moteus::PositionMode::Command cmd = {};
  cmd.accel_limit = 6.0;

  auto print = [](auto maybe_result) {
    if (maybe_result) {
      const auto r = maybe_result->values;
      const uint16_t reg = r.extra[0].register_number;
      if (reg == kFirmwareVersion) {
        const auto int_version = static_cast<int>(r.extra[0].value);
        ::printf("Firmware version 0x%x\n", int_version);
      } else {
        ::printf("%3d p/v/t=(%7.3f,%7.3f,%7.3f)  v/t/f=(%5.1f,%5.1f,%3d)   \r",
                 static_cast<int>(r.mode), r.position, r.velocity, r.torque, r.voltage, r.temperature,
                 r.fault);
        ::fflush(stdout);
      }

      if (r.mode == moteus::Mode::kFault || r.mode == moteus::Mode::kPositionTimeout) {
        ::printf("\n\nExiting because of fault %d/%d.\n", static_cast<int>(r.mode), r.fault);
        std::exit(1);
      }
    }
  };

  moteus::Query::Format custom_fmt;
  /* Initially, only care about the mode/fault for the initial position request
   * while requesting the FW version
   */
  custom_fmt.mode = moteus::kInt8;
  custom_fmt.position = moteus::kIgnore;
  custom_fmt.velocity = moteus::kIgnore;
  custom_fmt.torque = moteus::kIgnore;
  custom_fmt.voltage = moteus::kIgnore;
  custom_fmt.temperature = moteus::kIgnore;
  custom_fmt.fault = moteus::kInt8;
  custom_fmt.extra[0].register_number = kFirmwareVersion;
  custom_fmt.extra[0].resolution = moteus::kInt32;

  while (true) {
    cmd.position = pos_max - 0.2;
    print(controller.SetPositionWaitComplete(cmd, 0.001, nullptr, &custom_fmt));

    /* We want to ask the Firmware version once */
    if (custom_fmt.extra[0].register_number == kFirmwareVersion) {
      /* Restore the default queries for now */
      custom_fmt = {};
      custom_fmt.mode = moteus::kInt8;
      custom_fmt.position = moteus::kIgnore;
      custom_fmt.velocity = moteus::kIgnore;
      custom_fmt.torque = moteus::kIgnore;
      custom_fmt.voltage = moteus::kIgnore;
      custom_fmt.temperature = moteus::kIgnore;
      custom_fmt.fault = moteus::kInt8;
    }

    cmd.position = pos_min + 0.2;
    print(controller.SetPositionWaitComplete(cmd, 0.001, nullptr, &custom_fmt));
  }

  return 0;
}
```

```
  can1  00008001  [24]  01 00 0A 0D 20 CD CC 4C 3F 0D 29 00 00 C0 40 11 00 11 0B 11 0F 19 81 02
  can1       100  [16]  21 00 0A 21 0B 01 21 0F 00 29 81 02 0B 01 00 00
  can1  00008001  [21]  01 00 0A 0D 20 CD CC 4C BF 0D 29 00 00 C0 40 11 00 11 0B 11 0F
  can1       100  [16]  21 00 0A 21 0B 01 21 0F 00 29 81 02 0B 01 00 00
```

jpieper, 2025-06-26:
Want to try this patch?

```
diff --git a/lib/cpp/mjbots/moteus/moteus_transport.h b/lib/cpp/mjbots/moteus/moteus_transport.h
index dd45e2cd..95dcf421 100644
--- a/lib/cpp/mjbots/moteus/moteus_transport.h
+++ b/lib/cpp/mjbots/moteus/moteus_transport.h
@@ -285,6 +285,18 @@ class TimeoutTransport : public Transport {
   }
 
 
+  static size_t RoundUpDlc(size_t size) {
+    if (size <= 8) { return size; }
+    if (size <= 12) { return 12; }
+    if (size <= 16) { return 16; }
+    if (size <= 20) { return 20; }
+    if (size <= 24) { return 24; }
+    if (size <= 32) { return 32; }
+    if (size <= 48) { return 48; }
+    if (size <= 64) { return 64; }
+    return size;
+  }
+
  protected:
   virtual int CHILD_GetReadFd() const = 0;
   virtual void CHILD_SendCanFdFrame(const CanFdFrame&) = 0;
@@ -708,18 +720,6 @@ class Fdcanusb : public details::TimeoutTransport {
     tx_buffer_size_ = 0;
   }
 
-  static size_t RoundUpDlc(size_t size) {
-    if (size <= 8) { return size; }
-    if (size <= 12) { return 12; }
-    if (size <= 16) { return 16; }
-    if (size <= 20) { return 20; }
-    if (size <= 24) { return 24; }
-    if (size <= 32) { return 32; }
-    if (size <= 48) { return 48; }
-    if (size <= 64) { return 64; }
-    return size;
-  }
-
   static int ParseHexNybble(char c) {
     if (c >= '0' && c <= '9') { return c - '0'; }
     if (c >= 'a' && c <= 'f') { return c - 'a' + 10; }
@@ -828,8 +828,11 @@ class Socketcan : public details::TimeoutTransport {
       // Set the frame format flag if we need an extended ID.
       send_frame.can_id |= (1 << 31);
     }
-    send_frame.len = frame.size;
+    send_frame.len = RoundUpDlc(frame.size);
     std::memcpy(send_frame.data, frame.data, frame.size);
+    if (send_frame.len != frame.size) {
+      std::memset(&send_frame.data[frame.size], 0x50, send_frame.len - frame.size);
+    }
 
     using F = CanFdFrame;
 
```

jbtheou, 2025-06-26:
Works great! The issue is fixed and I am properly seeing the padding. Thanks a lot for the quick feedback here, very much appreciated

```
  can1  00008001  [24]  01 00 0A 0D 20 CD CC 4C 3F 0D 29 00 00 C0 40 11 00 11 0B 11 0F 19 81 02
  can1       100  [16]  21 00 0A 21 0B 01 21 0F 00 29 81 02 0B 01 00 00
  can1  00008001  [24]  01 00 0A 0D 20 CD CC 4C BF 0D 29 00 00 C0 40 11 00 11 0B 11 0F 50 50 50
  can1       100  [12]  21 00 0A 21 0B 01 21 0F 00 50 50 50
```

closed 2025-06-26 by jpieper