mjbots/moteus #95: How to set GPIO mode through CAN

Elijah884, 2025-08-08:
I am trying to set the AUX 1 GPIO pins the software quadrature input (mode 4) but I cannot figure out how to do it over CAN. All of my other commands work so far, so I know my set up is good, but I think I just don't understand the GPIO registers properly. I am trying to write a value of 4 to register 0x05C (which I'm assuming is incorrect) using "cangen can0 -e -I 0x8001 -D 015C04 -f", but when I run "cangen can0 -e -I 0x8001 -D 115C -f" to read the value at 5C it always just returns 00. Any help on this would be much appreciated, thank you!

Elijah884, 2025-08-09:
I'm using the Waveshare 2-channel CAN FD hat

Elijah884, 2025-08-09:
Here's what I ran and what was returned:
'root@shuttle1:~# python3 -m moteus_gui.tview --can-disable-brs
qt.pysideplugin: No python files found in '/usr/local/lib/python3.12/dist-packages/moteus_gui'.
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
^X^C'

Here's the candump for can0 from this command:
'can0  00008001  [03]  42 01 30
  can0       100  [03]  41 01 00
  can0  00008001  [03]  42 01 30
  can0       100  [03]  41 01 00
  can0  00008001  [03]  42 01 30
  can0       100  [03]  41 01 00
  can0  00008001  [03]  42 01 30
  can0       100  [03]  41 01 00
  can0  00008001  [03]  42 01 30
  can0       100  [03]  41 01 00
  can0  00008001  [03]  42 01 30
  can0       100  [03]  41 01 00
  can0  00008001  [03]  42 01 30
  can0       100  [03]  41 01 00
  can0  00008001  [03]  42 01 30
  can0       100  [03]  41 01 00'

It just sends those same messages back and forth forever.

jpieper, 2025-08-10:
As a graphical program it definitely won't work over ssh with x forwarding and maybe not with either.

I assume you have a graphical desktop on your pi? Can you share a screenshot or photo when attempting to run it on the pi directly?

Elijah884, 2025-08-11:
Sorry for such a late reply but I should be able to get a screenshot soon

Elijah884, 2025-08-11:
So apparently I installed the pi OS version without a desktop so I can't open the graphics window, but your earlier suggestion to use the python script to generate the diagnostic commands is working so I'll probably just use that. Thank you for your help!

Elijah884, 2025-08-11:
Following the script from earlier which creates the diagnostic command in python, how do I return the telemetry to normal after running await s.write_message(b'tel stop')?

Elijah884, 2025-08-14:
Never mind, it turns out I just wasn't closing the socketcan bus properly and it was messing some stuff up