mjbots/moteus #97: GPIO pins won't enable

Elijah884, 2025-08-15:
Hello, 

Whenever I try to enable the GPIO pins (either through register commands or diagnostic commands) they always remain in an off state. Blwo I have the output of gpio.py from the examples section of the moteus library, which shows all pins on the GPIO are set to 0:
root@shuttle1:~# python3 gpio_test.py
AUX1
  Pin 0 - 0
  Pin 1 - 0
  Pin 2 - 0
  Pin 3 - 0
  Pin 4 - 0

AUX2
  Pin 0 - 0
  Pin 1 - 0
  Pin 2 - 0
  Pin 3 - 0

From moteus.Controller.query()
AUX1
  Pin 0 - 0
  Pin 1 - 0
  Pin 2 - 0
  Pin 3 - 0
  Pin 4 - 0

AUX2
  Pin 0 - 0
  Pin 1 - 0
  Pin 2 - 0
  Pin 3 - 0
SocketcanBus was not properly shut down

Additionally, if I try to send a command to 5C to set it to 7F (enabling all gpio pins) the controller doesn't respond to that command and then states that all gpio pins are disabled when I read from 5C:
Sent:  00 01 5C 7F
Received:
Sent:  10 01 5C
Received:  20 01 5C 00

I'm not sure if I'm misinterpreting this or if there's some other setting I need to change first before enabling GPIO, but I've tried a few configurations and can't get it to work.

Thank you for your help!

jpieper, 2025-08-15:
I think error 12 means you have a SPI MA600 configured on the aux port but configuration was not successful. Can you share either the config dump, or it is easier if you can share a screenshot from tview with the config and telemetry.

jpieper, 2025-08-15:
And I should say, if there is any error, then nothing will work on that aux port. Thus you need to understand and resolve the error.

Elijah884, 2025-08-15:
Makes sense, I have pictures of the config and telemetry screens, sorry they aren't screenshots but scrot, print screen, and gnome screenshot all decided to not work

[image]
[image]

[image]
[image]

Elijah884, 2025-08-15:
I'm going to try it on Aux2 instead to see if I get that same error

jpieper, 2025-08-15:
What type of controller do you have?  I.e. is it a moteus-x1?  You have onboard MA600 configured on aux1, but only the moteus-x1 has an onboard MA600.

Elijah884, 2025-08-15:
I'm not sure what it means but in the telemetry it says <error:quad_pin_error:6>. Where do I find what these aux errors mean? I can't find them in the documentation anywhere.

jpieper, 2025-08-15:
Most of the aux errors mean you do not have compatible pins configured for the devices you have configured.  The quad pin error means that you do not have appropriate pins configured to be used for quadrature, but attempted to enable quadrature.

If you want to use quadrature input you need to first:

aux2.quadrature.enable = 1

Then configure which pins you want.  For instance, if you want to use the hardware enabled quadrature pins from: https://github.com/mjbots/moteus/blob/main/docs/reference.md#aux2--abs  you would set:

aux2.pins.2.mode 5  # hw quad
aux2.pins.3.mode 5  # hw quad

I will note that in your screenshots earlier your aux1.error was actually error 1, not error 12.  That is a SPI pin error.  I'm not entirely sure why you would have a SPI pin error when the onboard MA600 is configured.  However, an error 12 would easily happen if you had connected external circuits to the aux1 pins but had not disabled the MA600 by setting 

aux1.spi.mode = 1 # disabled

All of this is of course much easier in tview where the various enumerations and status codes have their text strings shown.

You can learn about the pin and function configuration process by watching this video: https://www.youtube.com/watch?v=a3UA6yzzgkE  or reading this post: https://blog.mjbots.com/2022/06/27/flexible-i-o-auxiliary-port-configuration/

jpieper, 2025-08-19:
Have you ever calibrated a moteus controller successfully before?

Can you share a screenshot of the drv8323 tree?